zsmartsystems/com.zsmartsystems.zigbee #1281: ZStack-Driver Branch Compilation

Aimen-Hammou, 2021-11-30:
Hi,

I would like to test the zstack-driver branch ( special thanks to @Hedda and @t-8ch ) of the repository, but when I run the `mvn clean install` it runs into some problems.

Below you can find the log. 

The main repository works just fine, just have to make sure eclipse opens it the right way.

(Would also like to thanks @cdjackson for the awesome job! )

Log:
`
[ERROR] 
[ERROR] Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 9, column 10
[FATAL] Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack.autocode:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 14, column 13
[FATAL] Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.console.zstack:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 9, column 13
 @
[ERROR] The build could not read 3 projects -> [Help 1]
[ERROR]
[ERROR]   The project com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack:[unknown-version] (C:\Users\Hi Acer\Desktop\ZigBee4JAVA\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee.dongle.zstack\pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 9, column 10 -> [Help 2]
[ERROR]
[ERROR]   The project com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack.autocode:[unknown-version] (C:\Users\Hi Acer\Desktop\ZigBee4JAVA\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee.dongle.zstack.autocode\pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.dongle.zstack.autocode:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 14, column 13 -> [Help 2]
[ERROR]
[ERROR]   The project com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.console.zstack:[unknown-version] (C:\Users\Hi Acer\Desktop\ZigBee4JAVA\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee.console.zstack\pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for com.zsmartsystems.zigbee:com.zsmartsystems.zigbee.console.zstack:[unknown-version]: Could not find artifact com.zsmartsystems:zigbee:pom:1.1.11-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 9, column 13 -> [Help 2]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
[ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/UnresolvableModelException`

_PS. In case I made any mistake in the way I formatted the ticket please let me know, so I can fix it._

t-8ch, 2021-11-30:
@Aimen-Hammou From which repo did you get the code? You should use `zstack_driver` from https://github.com/t-8ch/com.zsmartsystems.zigbee/tree/zstack_driver .

Longer verbatim text should be enclosed in lines of triple-backticks. ("```")

Aimen-Hammou, 2021-11-30:
@t-8ch Thank you, tomorrow I will try.

Do you mind sharing the main class where you tested your dongle? Or how to initialize it and connect to it via serial port?

Would like to test CC2562 and try to connect to my thermostat

(Also thank you for your correction, I've learned something new)

Aimen-Hammou, 2021-12-01:
@t-8ch I've tried your fork but gives me some errors regarding missing license header, right at the ZssBee. I added `-Dlicense.skip=true` to it and skip the first block of errors, but fails at the dongle.xbee (see below for log) 

I run mvn clean install in the main folder.

```
 Reactor Summary for ZssBee 1.4.3-SNAPSHOT:
[INFO]
[INFO] ZssBee ............................................. SUCCESS [  3.015 s]
[INFO] com.zsmartsystems.zigbee ........................... SUCCESS [01:39 min]
[INFO] com.zsmartsystems.zigbee.autocode .................. SUCCESS [  4.638 s]
[INFO] com.zsmartsystems.zigbee.dongle.xbee ............... FAILURE [  1.815 s]
[INFO] com.zsmartsystems.zigbee.dongle.xbee.autocode ...... SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.cc2531 ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.ember .............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.ember.autocode ..... SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.conbee ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.telegesis .......... SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.telegesis.autocode . SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.zstack ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.zstack.autocode .... SKIPPED
[INFO] com.zsmartsystems.zigbee.console ................... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.ember ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.serial .................... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.telegesis ......... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.zstack ............ SKIPPED
[INFO] com.zsmartsystems.zigbee.console.main .............. SKIPPED
[INFO] com.zsmartsystems.zigbee.test ...................... SKIPPED
[INFO] com.zsmartsystems.zigbee.p2repo .................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:50 min
[INFO] Finished at: 2021-12-01T09:36:17+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.0:testCompile (default-testCompile) on project com.zsmartsystems.zigbee.dongle.xbee: Compilation failure
[ERROR] /C:/Users/Hi Acer/Desktop/ZigBee4JAVA/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee.dongle.xbee/src/test/java/com/zsmartsystems/zigbee/dongle/xbee/internal/XBeeFrameHandlerTest.java:[101,5] com.zsmartsystems.zigbee.dongle.xbee.internal.XBeeFrameHandlerTest.TestPort is not abstract and does not override abstract method setRts(boolean) in com.zsmartsystems.zigbee.transport.ZigBeePort
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :com.zsmartsystems.zigbee.dongle.xbee
```

**Side Note**
 When compiling with NetBeans I encouter this problem
```
Plugin org.apache.maven.plugins:maven-resources-plugin:2.5 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.apache.maven.plugins:maven-resources-plugin:jar:2.5: Could not transfer artifact org.apache.maven.plugins:maven-resources-plugin:pom:2.5 from/to central (http://repo.maven.apache.org/maven2): Failed to transfer file: http://repo.maven.apache.org/maven2/org/apache/maven/plugins/maven-resources-plugin/2.5/maven-resources-plugin-2.5.pom. Return code is: 501 , ReasonPhrase:HTTPS Required. -> [Help 1]
```
If I change from http to https it should resolve this error?

t-8ch, 2021-12-01:
The compile error is now fixed in the latest version on GitHub.
About NetBeans I have no idea.

Aimen-Hammou, 2021-12-01:
I've tested the z-stack driver branch of your fork.

Xbee now works but CC2531 gives me an error (regarding your last commit to that branch maybe?)

```
[INFO]
[INFO] ZssBee ............................................. SUCCESS [  3.220 s]
[INFO] com.zsmartsystems.zigbee ........................... SUCCESS [01:37 min]
[INFO] com.zsmartsystems.zigbee.autocode .................. SUCCESS [  4.478 s]
[INFO] com.zsmartsystems.zigbee.dongle.xbee ............... SUCCESS [ 11.791 s]
[INFO] com.zsmartsystems.zigbee.dongle.xbee.autocode ...... SUCCESS [  2.541 s]
[INFO] com.zsmartsystems.zigbee.dongle.cc2531 ............. FAILURE [  1.802 s]
[INFO] com.zsmartsystems.zigbee.dongle.ember .............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.ember.autocode ..... SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.conbee ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.telegesis .......... SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.telegesis.autocode . SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.zstack ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.dongle.zstack.autocode .... SKIPPED
[INFO] com.zsmartsystems.zigbee.console ................... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.ember ............. SKIPPED
[INFO] com.zsmartsystems.zigbee.serial .................... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.telegesis ......... SKIPPED
[INFO] com.zsmartsystems.zigbee.console.zstack ............ SKIPPED
[INFO] com.zsmartsystems.zigbee.console.main .............. SKIPPED
[INFO] com.zsmartsystems.zigbee.test ...................... SKIPPED
[INFO] com.zsmartsystems.zigbee.p2repo .................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:02 min
[INFO] Finished at: 2021-12-01T10:30:18+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.0:testCompile (default-testCompile) on project com.zsmartsystems.zigbee.dongle.cc2531: Compilation failure
[ERROR] /C:/Users/Hi Acer/Desktop/ZigBee4JAVA/com.zsmartsystems.zigbee-zstack_driver/com.zsmartsystems.zigbee.dongle.cc2531/src/test/java/com/zsmartsystems/zigbee/dongle/cc2531/Cc2351TestPacket.java:[61,5] com.zsmartsystems.zigbee.dongle.cc2531.Cc2351TestPacket.TestPort is not abstract and does not override abstract method setRts(boolean) in com.zsmartsystems.zigbee.transport.ZigBeePort
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :com.zsmartsystems.zigbee.dongle.cc2531
```

t-8ch, 2021-12-01:
And again

Aimen-Hammou, 2021-12-01:
Great>! It now compiles just fine. (Doesn't give me that missing license header anymore, so I don't have to add the flag to mvn)

Do you mind sharing how you tested your dongle? I wanted to test mine by using a java main class, but don't really know how.

t-8ch, 2021-12-01:
I built a version of the OpenHab binding using the new version of this library and tested with OpenHab.
As I said before, at the current state testing by endusers is of very limited utility.

Aimen-Hammou, 2021-12-01:
Thanks! I will close this issue.

closed 2021-12-01 by Aimen-Hammou

Hedda, 2021-12-01:
@t-8ch @Aimen-Hammou Perhaps keep an ongoing discussion for end-users in openHAB forum for testing and progress reporting? -> https://community.openhab.org/t/discussion-about-ti-z-stack-3-x-and-cc2652-and-cc1352-zigbee-coordinator-adapters-with-openhab-zigbee-binding/129385