zsmartsystems/com.zsmartsystems.zigbee #1290: ConcurrentModificationException in ZigBeeTransactionManager.sendNextTransaction

TomTravaglino, 2022-01-17:
This Exception is thrown when iterating over `outstandingQueues`
```
java.util.ConcurrentModificationException: null
at java.util.ArrayList$Itr.checkForComodification(checkForComodification:911)
at java.util.ArrayList$Itr.next(next:861)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(sendNextTransaction:661)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.transactionComplete(transactionComplete:540)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.cancel(cancel:357)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue.transactionComplete(transactionComplete:321)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.transactionComplete(transactionComplete:526)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.failTransaction(failTransaction:446)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.transactionStatusReceived(transactionStatusReceived:491)
at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager$3.run(run:579)
at java.util.concurrent.ThreadPoolExecutor.runWorker(runWorker:1149)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(run:624)
at java.lang.Thread.run(run:748)utor$Worker.run(run:624)
at java.lang.Thread.run(run:748)
```

`outstandingQueues` is a simple `ArrayList`. Changing it to a `CopyOnWriteArrayList` should fix it. WDYT?

TomTravaglino, 2022-01-18:
@cdjackson 
I created a PR for this.

cdjackson, 2022-01-18:
Thanks @TomTravaglino . Sorry - I missed this yesterday but I think this is fine.

PR #1291 (2022-01-18): Use thread-safe list type for outstandingQueues
Resolves #1290 

This PR fixes an issue where a `ConcurrentModificationException` rises when iterating over `outstandingQueues`. `outstandingQueues` is now an instance of `CopyOnWriteArrayList` instead of an `ArrayList`.
closed 2022-01-18 by cdjackson