zsmartsystems/com.zsmartsystems.zigbee #1301: PTM215Z support - Error creating instance of EzspFrame

RandomReaper, 2022-02-09:
**Describe the bug**
openHAB 3.3.0.M1 + zigbee binding

**To Reproduce**
click a button on the PTM215Z

**Expected behavior**
Receive a event

**Additional context**
zigbee dongle : POPP zb-stick (ember with reported firmware version : 6.10.0.0)

```
2022-02-09 11:14:39.608 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=7, reTx=false, data=DE 90 01 C5 00 7F D7 53 00 14 8A 70 01 14 8A 70 01 54 02 01 00 00 53 07 00 00 64 34 87 07 3B FF 00]
2022-02-09 11:14:39.609 [DEBUG] [s.zigbee.dongle.ember.ezsp.EzspFrame] - Error creating instance of EzspFrame
java.lang.reflect.InvocationTargetException: null
	at jdk.internal.reflect.GeneratedConstructorAccessor151.newInstance(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:490) ~[?:?]
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.EzspFrame.createHandler(EzspFrame.java:491) [bundleFile:?]
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler$AshReceiveParserThread.run(AshFrameHandler.java:309) [bundleFile:?]
Caused by: java.lang.ArrayIndexOutOfBoundsException: Index 33 out of bounds for length 33
	at com.zsmartsystems.zigbee.dongle.ember.internal.serializer.EzspDeserializer.deserializeUInt32(EzspDeserializer.java:163) ~[?:?]
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.command.EzspGpepIncomingMessageHandler.<init>(EzspGpepIncomingMessageHandler.java:139) ~[?:?]
	... 5 more
2022-02-09 11:14:39.611 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: No frame handler created for AshFrameData [frmNum=3, ackNum=7, reTx=false, data=DE 90 01 C5 00 7F D7 53 00 14 8A 70 01 14 8A 70 01 54 02 01 00 00 53 07 00 00 64 34 87 07 3B FF 00]
2022-02-09 11:14:39.611 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=4, notRdy=false]
2022-02-09 11:14:39.836 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=4, ackNum=7, reTx=false, data=DE 90 01 C5 00 7F D7 54 00 14 8A 70 01 14 8A 70 01 54 02 01 00 00 54 07 00 00 65 2C 15 E8 75 FF 00]
2022-02-09 11:14:39.836 [DEBUG] [s.zigbee.dongle.ember.ezsp.EzspFrame] - Error creating instance of EzspFrame
java.lang.reflect.InvocationTargetException: null
	at jdk.internal.reflect.GeneratedConstructorAccessor151.newInstance(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:490) ~[?:?]
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.EzspFrame.createHandler(EzspFrame.java:491) [bundleFile:?]
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler$AshReceiveParserThread.run(AshFrameHandler.java:309) [bundleFile:?]
Caused by: java.lang.ArrayIndexOutOfBoundsException: Index 33 out of bounds for length 33
	at com.zsmartsystems.zigbee.dongle.ember.internal.serializer.EzspDeserializer.deserializeUInt32(EzspDeserializer.java:163) ~[?:?]
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.command.EzspGpepIncomingMessageHandler.<init>(EzspGpepIncomingMessageHandler.java:139) ~[?:?]
	... 5 more
2022-02-09 11:14:39.839 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: No frame handler created for AshFrameData [frmNum=4, ackNum=7, reTx=false, data=DE 90 01 C5 00 7F D7 54 00 14 8A 70 01 14 8A 70 01 54 02 01 00 00 54 07 00 00 65 2C 15 E8 75 FF 00]
2022-02-09 11:14:39.839 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=5, notRdy=false]
2022-02-09 11:14:40.248 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
```

RandomReaper, 2022-02-11:
The impact on OH is no event coming from that zigbee device.

cdjackson, 2022-02-11:
This framework, and OH especially doesn’t support the GPD commands at this time, so this command is not used in OH, so it will have no impact on OH when this is fixed.

cdjackson, 2022-02-19:
Silabs advise that the GPAddress is a union, not a flat structure as defined in the documentation. The C file shows this -:

```
typedef struct {
  union {
    /** The IEEE address is used when the application identifier is
     *  ::EMBER_GP_APPLICATION_IEEE_ADDRESS.
     */
    EmberEUI64 gpdIeeeAddress;
    /** The 32-bit source identifier is used when the application identifier is
     *  ::EMBER_GP_APPLICATION_SOURCE_ID.
     *
     */
    EmberGpSourceId sourceId;
  } id;
  /** Application identifier of the GPD. */
  EmberGpApplicationId applicationId;
  uint8_t endpoint;
} EmberGpAddress;
```

I'm still not sure that this allows the GP frames to be decoded correctly, but it should eliminate the exception since the frame length is now correct. I'm continuing to discuss this with Silabs.

PR #1303 (2022-02-22): Fix gpepIncomingMessageHandler definition
Silabs advise that the `EmberGPAddress` is a union, not a flat structure as defined in the documentation. The C file shows this -:

```
typedef struct {
  union {
    /** The IEEE address is used when the application identifier is
     *  ::EMBER_GP_APPLICATION_IEEE_ADDRESS.
     */
    EmberEUI64 gpdIeeeAddress;
    /** The 32-bit source identifier is used when the application identifier is
     *  ::EMBER_GP_APPLICATION_SOURCE_ID.
     *
     */
    EmberGpSourceId sourceId;
  } id;
  /** Application identifier of the GPD. */
  EmberGpApplicationId a [... cut]
closed 2022-02-22 by cdjackson