zsmartsystems/com.zsmartsystems.zigbee #1305: EZSP ASH handler is not always closed.

cdjackson, 2022-02-26:
If the ASH handler fails to initialise correctly, then it doesn't get closed when the Ember driver shuts down. A flag was introduced in #1287 to only perform the shutdown after it was configured and this is set at the end of the `initialize` method. If there is a failure  in the `initialize` method, then the flag isn't set, and then when `shutdown` is called, the `frameHandler` is not closed - leaving the serial port open. In the log we see a continual `ASH: Reconnect` being loged in the `AshFrameHandler`.

@TomTravaglino just pinging you here as I recall we had quite some discussion about this. I've not looked at this in detail yet, but I guess that `shutdown` needs to reverse `initialize` - so if the port is open, we should close it, and if there is a handler open, it should be closed...

TomTravaglino, 2022-03-01:
Hello @cdjackson. It took a look at it and yes, `shutdown` exits immediately if the `isConfigured` flag is not `true`. I guess we need to remove the check and let `shutdown` do its job.

TomTravaglino, 2022-03-01:
@cdjackson I created a PR to suggest a solution. Please take a look.

PR #1311 (2022-03-01): Changed shutdown methods of drivers to always close handlers
Resolves #1305 

This PR fixes the issue where `shutdown` doesn't close the handlers when the initialization was not successful.
closed 2022-03-01 by cdjackson