zsmartsystems/com.zsmartsystems.zigbee #1332: EZSP v9 (EmberZNet Serial Protocol version 9) for Silicon Labs

Hedda, 2022-06-10:
**Is your feature request related to a problem? Please describe.**

Request support for EZSP v9 (EmberZNet Serial Protocol version 9) for Silicon Labs.

**Describe the solution you'd like**

Silicon Labs Gecko SDK 4.1.0 (GSDK 4.1) was released a couple of days ago and with it Zigbee EmberZNet SDK 7.1.0.0

https://github.com/SiliconLabs/gecko_sdk/releases

Importantly this EmberZNet SDK 7.1.0 added an update to EZSP v9 (EmberZNet Serial Protocol version 9) which might break this if a user upgrades to so Silabs Zigbee EmberZNet SDK 7.1.0.0 NCP firmware.

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf

That is unless EZSP adapter implementation is updated to support additions to EZSP v9. See EZSP Reverence guide Rev 4.4 and 4.3

zigbee-herdsman developers please EZSP Reference Guide for 7.0.0 with the changes from 6.10.x highlighted is here:

https://web.archive.org/web/20220401230237/https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

and for 7.1.0 with the changes from 7.0.0 > 7.1.0 hightlighted is here (until the next version):

https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

**Describe alternatives you've considered**

Downgrade firmware or possibly use Silabs Zigbee EmberZNet SDK 7.1.0.0 NCP firmware in backwards compatibility mode if possible.

**Additional context**

cdjackson, 2022-06-10:
I am aware of this update. In general the protocol is largely the same, but there is a breaking change in some responses that needs further though on how to handle this as it may signal a significant change by SL.

For now I wouldn't suggest using the latest 7.1 firmware. 7.0.2 is fine.

Hedda, 2022-06-13:
OK that is cool.

Please however take note that because of its new (Alpha) support for concurrently running firmware with Zigbee NCP and OpenThread RPC at the same time announced in Silicon Labs Gecko SDK v4.1.0 I do believe that sooner or later going with EmberZNet 7.1.x firmware or even in the relatively near future for early adopters will be tempting to both manufacturers of Zigbee radio USB adapters and end-users owning existing Silicon Labs EFR32MG12, EFR32MG13, and EFR32MG21 adapters.

As such firmware will mean that could use the same EFR32MGxx adapter as a Zigbee Coordinator for existing Zigbee devices and an OpenThread Border Router (OTBR) for the upcoming Matter over Thread (formerly Project CHIP / Connected Home over IP). So in essence users could just flash their existing Silabs Zigbee Coordinator and have it also act as a radio for Matter over Thread.

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.1.0

<html>
<body>

Component | Version | Summary | Details
-- | -- | -- | --
[OpenThread SDK](https://www.silabs.com/wireless/thread#software) | 2.1.0.0 | - Alpha SPI support for OpenThread RCP without CPC<br> - Thread 1.2 and 1.3 support for OpenThread<br> - Updated GCC compiler version to 10.3.1<br> - **Alpha Concurrent Multiprotocol Zigbee in NCP mode and OpenThread in RCP mode**<br> - Alpha Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode | [Release Notes](https://www.silabs.com/documents/public/release-notes/open-thread-release-notes-2.1.0.0.pdf)
[Zigbee EmberZNet SDK](https://www.silabs.com/wireless/zigbee#software) | 7.1.0.0 | - 2.4GHz Zigbee Smart Energy support for xG24<br> - 802.15.4 Signal Identifier and MAC CCA Mode 2 and 3 support for xG24<br> - Zigbee Green Power Gateway Backup<br> - Updated GCC and IAR compiler version<br> - **Alpha Concurrent Multiprotocol Zigbee in NCP mode and Open-Thread in RCP mode**<br> - Alpha Dynamic Multiprotocol Bluetooth and multi-PAN 802.15.4 in RCP mode | [Release Notes](https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.1.0.0.pdf)

</body>
</html>

cdjackson, 2022-06-13:
Sure - I'm well aware of this, and well aware of the changes that you list here from the release notes, but so far this is not a priority. It's also not hard to implement the changes, but there's no need to jump into this as you seem to be suggesting. I support many commercial systems here and this is as yet not a driver.

Hedda, 2022-09-18:
FYI, there's now also Zigbee EmberZNet 7.1.x NCP firmware images with EZSP v9 available from community firmware developers:

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

https://github.com/grobasoz/zigbee-firmware/tree/master/EFR32MG2x-768k

So specifically for OpenHAB ZigBee Binding, it might be good to be aware some users might have flashed newer those firmware.

Community available Zigbee USB adapters hardware that supports that firmware is ITead's new EFR32MG21 based “ZBDongle-E” USB adapter (a.k.a. “Sonoff Zigbee 3.0 USB Dongle Plus V2”) which has just begun to be popular as is those “ZBDongle-E” dongles are now starting to be available globally, as well as the newer EFR32MG21 based "ZB-GW04" USB stick (a.k.a. "easyiot Zigbee dongle") that in turn uses the "SM-011" radio module (ZYZBP008) by CoolKit Technologies which has previously also used in some other Zigbee gateway products from China.

https://itead.cc/product/zigbee-3-0-usb-dongle/

https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20220917212336&origin=y&SearchText=ZB-GW04+

https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20220917212448&SearchText=easyiot+efr32mg21&spm=a2g0o.productlist.1000002.0

https://www.aliexpress.com/item/1005003595899955.html

https://www.aliexpress.com/item/1005003595793997.html

https://www.coolkit.cn/product/sm-011/

https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee

PS: EZSP v9 support is currently being tested by developers with zigpy + bellows library based applications such as Home Assistant ZHA integration, Zigbee Plugin for Domoticz, and Zigbee Plugin for Jeedom, as well as the zigbee-herdsman based Zigbee2MQTT and IoBroker Zigbee implementations.

Hedda, 2023-02-05:
FYI, Nabu Casa is now offering official EmberZNet 7.1.x.x (EZSP v9) firmware images for their Home Assistant SkyConnect:

https://www.home-assistant.io/skyconnect/

https://github.com/NabuCasa/silabs-firmware/tree/main/EmberZNet

xsp1989 is also offering official EmberZNet 7.1.x.x and EmberZNet 7.2.x.x for his ZB-GW04 USB stick (a.k.a. "easyiot Zigbee dongle") that he designed based on "SM-011 V1.x" (ZYZBP008) radio module from CoolKit-Technologies (which in turn has an EFR32MG21 SoC). 

https://www.aliexpress.com/item/1005002791666029.html

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

sraillard, 2023-02-14:
I had a look to their new GitHub repository and we can notice that the EZSP protocol is incremented each time a new Gecko SDK is released... Latest version is 10, whereas UG100 still mention version 9, like usual the documentation isn't updated.

Gecko SDK v4.0:
- https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.0/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
- EZSP_PROTOCOL_VERSION = 8

Gecko SDK v4.1:
- https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.1/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
- EZSP_PROTOCOL_VERSION = 9

Gecko SDK v4.2:
- https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.2/protocol/zigbee/app/util/ezsp/ezsp-protocol.h
- EZSP_PROTOCOL_VERSION = 10

It doesn't look like there are much differences between these protocol definitions. Silabs really has an issue in documentation their changes and not speaking of providing a kind of long term support version of EmberZNet...

Hedda, 2023-02-19:
The only major addition in EZSP v10 looks to be incomingNetworkStatusHandler, which was previously changeSourceRouteHandler so need to handle that it was renamed in that version of the protocol.

cdjackson, 2023-02-26:
> Silabs really has an issue in documentating their changes

Yes, it is normally quite poor. Unfortunately there are often undocumented changes as well, and I've sought clarifications from Silabs a few years ago that they agreed were incorrectly documented, and they are still wrong :(

sraillard, 2023-03-20:
@cdjackson, I have seen your commit for using EZSP up to version 10.
Meanwhile, silabs has released EmberZNet 7.2.2.0 with EZSP version 11, here is an extract from the release notes:
[image]

Hedda, 2023-03-20:
Indeed, EZSP protocol version has once again been incremented to EZSP v11 in Gecko SDK v4.2 / Zigbee EmberZNet v7.2 and once again that EZSP v11 was not mentioned in the latest UG100 (EZSP Reference Guide) documentation from Silicon Labs which SiLabs still has only updated for EZSP v9 and state that no additions were added between EmberZNet PRO Release 7.1.1 or 7.2.2:

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.2/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

EZSP v11 version was only mentioned under miscellaneous changes in release Zigbee EmberZNet SDK 7.2.2.0 GA (Gecko SDK Suite 4.2 March 8, 2023):

"Updated the ezspPollHandler function with updated input arguments, that required updating the EZSP_PROTOCOL_VERSION to 0x0B"

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.2.0.pdf

0x0B hexadecimal = 11 in decimal.

https://www.silabs.com/wireless/zigbee?tab=software#software

Hedda, 2023-04-17:
FYI, latest bug-fix version is now Silicon Labs Zigbee EmberZNet SDK 7.2.2.0 from Gecko SDK version 4.2.2.0 from March 10, 2023:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.2.2.0.pdf

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.2.2

closed 2023-04-22 by cdjackson

FranzSchi, 2023-09-01:
Hi everyone,

is there someone that was able to run Versions on a higer EZSP on Openhab?
I've tried in this order

Version 6.10.3.0, EZSP version 8  => Starting Version all fine.
Version 7.1.4.0 EZSP version 9 => Did not really work, my rollershutters got disconnected and i had no new devices in the "inbox"
Version 7.2.3.0 ESZP version 11 => Same as Version 9
Version 6.10.7.0, EZSP version 8  => Restored now backward with nvram reset and written again on it. Seems better till now.

I've tested it with and
   SonOff Zigbee Stick E
   ZB-GW04 V1.1

If I'm the only one then all fine.
If not then i could support here at least by providing log files.

/Franz

FranzSchi, 2023-09-01:
Hi @cdjackson,

it started, okay lets make it more complex.
Yes i had that error with Version 12 therefore i used version 9 and 11.

Version 7.3X.0 EZSP version 12 => Did not Start because it was version 12.
Version 6.10.3.0, EZSP version 8 => Starting Version all fine.
Version 7.1.4.0 EZSP version 9 => Did not really work, my rollershutters got disconnected and i had no new devices in the "inbox"
Version 7.2.3.0 ESZP version 11 => Same as Version 9
Version 6.10.7.0, EZSP version 8 => Restored now backward with nvram reset and written again on it. Seems better till now.

I've tested it with and
SonOff Zigbee Stick E
ZB-GW04 V1.1

/Franz

cdjackson, 2023-09-01:
> it started, okay lets make it more complex.

I don't really understand?

> Version 7.1.4.0 EZSP version 9 => Did not really work, my rollershutters got disconnected and i had no new devices in the "inbox"

Again, that's not an issue related to EZSP versions. It's really impossible to comment, and you talk about the inbox which is nothing to do with this library - it's probably best to discuss on the OH community support forum.

cdjackson, 2023-09-01:
From the logs you sent I can confirm there is no issue with the EZSP version. As stated already, if there was an issue with the EZSP version, the system would not start. Communications between the host and the NCP is working fine.