zsmartsystems/com.zsmartsystems.zigbee #1344: ASH processor thread not interrupted?

mikomarrache, 2022-09-06:
Hi Chris,

We recently had memory issues with the library where it looks like, at least according to the heap dump, that the processor threads accumulate in memory. I guess this issue only impacts applications that deal with multiple network managers. I do see the parser thread is interrupted when the `AshFrameHandler` is closed, but that's not the case for the processor thread.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/1938be0e8cf3aaac3198202355a653528b4443f2/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/ash/AshFrameHandler.java#L469

What do you think? Is that a bug or I am missing something?

Thanks!

cdjackson, 2022-09-11:
Hi Michael,

Sorry for the delay responding - I'm currently on holiday in Africa and internet access is a "little patchy".

I've not looked at this in detail, but I think that you are likely correct that this is a bug and the processor thread should also be interrupted when the handler is closed. As a matter of interest if you make this change does it resolve your issue? 

Please feel free to make a PR for this - I might not look too much at it for another week or two when I get back home, but it would be worth seeing if it resolves the leak.

mikomarrache, 2022-09-11:
Hi Chris,

I've created a PR:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1345

Thanks

cdjackson, 2022-10-01:
closed by #1345

closed 2022-10-01 by cdjackson