zsmartsystems/com.zsmartsystems.zigbee #1357: InitiateKeyEstablishmentRequestCommandSuccess(com.zsmartsystems.zigbee.app.seclient.ZclKeyEstablishmentServerTest) is unstable

leonschenk, 2022-11-02:
**Describe the bug**
InitiateKeyEstablishmentRequestCommandSuccess(com.zsmartsystems.zigbee.app.seclient.ZclKeyEstablishmentServerTest) is failing sometimes.

**To Reproduce**
mvn clean install (until the test fails)

**Expected behavior**
All tests pass each time

**Additional context**
I have not analyzed the issue

cdjackson, 2022-11-02:
I've just run this 30 times and it's not failed. Please can you at least provide some logs showing what the problem is.

leonschenk, 2022-11-02:
`ReadReportingConfigurationResponse [Basic: null -> null, cluster=0000, TID=--, records=[AttributeReportingStatusRecord [status=UNSUPPORTED_ATTRIBUTE, attributeIdentifier=0, direction=0]]]
Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 sec
Running com.zsmartsystems.zigbee.zcl.clusters.general.ReportAttributesCommandTest
ReportAttributesCommand [Basic: null -> null, cluster=0000, TID=--, reports=[AttributeReport [attributeDataType=UNSIGNED_16_BIT_INTEGER, attributeIdentifier=65285, attributeValue=860], AttributeReport [attributeDataType=FLOAT_32_BIT, attributeIdentifier=85, attributeValue=184.74998474121094]]]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 sec
Running com.zsmartsystems.zigbee.zcl.clusters.iaszone.ZoneEnrollRequestCommandTest
ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=63, commandId=1]
ZoneEnrollRequestCommand [IAS Zone: null -> null, cluster=0500, TID=--, zoneType=13, manufacturerCode=4174]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 sec
Running com.zsmartsystems.zigbee.zcl.clusters.iaszone.ZoneStatusChangeNotificationCommandTest
ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=7B, commandId=0]
ZoneStatusChangeNotificationCommand [IAS Zone: null -> null, cluster=0500, TID=--, zoneStatus=36, extendedStatus=0, zoneId=0, delay=0]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 sec
Running com.zsmartsystems.zigbee.zcl.clusters.onoff.OffWithEffectCommandTest
ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=12, commandId=64]
OffWithEffectCommand [On/Off: null -> null, cluster=0006, TID=--, effectIdentifier=0, effectVariant=0]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 sec
Running com.zsmartsystems.zigbee.zcl.clusters.otaupgrade.ImageNotifyCommandTest
ImageNotifyCommand [Ota Upgrade: 0000/1 -> DFF9/3, cluster=0019, TID=--, payloadType=0, queryJitter=72, manufacturerCode=4364, imageType=6, newFileVersion=16909063]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 sec
Running com.zsmartsystems.zigbee.zcl.clusters.otaupgrade.QueryNextImageCommandTest
QueryNextImageCommand [Ota Upgrade: null -> null, cluster=0019, TID=--, fieldControl=1, manufacturerCode=4364, imageType=98, fileVersion=16909571, hardwareVersion=17459]
QueryNextImageCommand [Ota Upgrade: null -> null, cluster=0019, TID=--, fieldControl=0, manufacturerCode=4364, imageType=98, fileVersion=16909571, hardwareVersion=null]
Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 sec
Running com.zsmartsystems.zigbee.zcl.clusters.scenes.AddScenesCommandTest
AddSceneCommand [Scenes: null -> null, cluster=0005, TID=--, groupId=1, sceneId=2, transitionTime=3, sceneName=Test Scene, extensionFieldSets=[ExtensionFieldSet [clusterId=0004, data=ByteArray [value=05 06]], ZclOnOffExtensionField [clusterId=6, onOff=true]]]   
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclColorControlClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclDiagnosticsClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclDoorLockClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclElectricalMeasurementClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.008 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclMultistateValueBasicClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclOccupancySensingClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclPollControlClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclPowerConfigurationClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclRssiLocationClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.013 sec
Running com.zsmartsystems.zigbee.zcl.clusters.ZclThermostatClusterTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.006 sec
Running com.zsmartsystems.zigbee.zcl.field.AttributeInformationTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 sec
Running com.zsmartsystems.zigbee.zcl.field.ByteArrayTest
ByteArray [value=01 02 03 04 05]
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 sec
Running com.zsmartsystems.zigbee.zcl.field.ReadAttributeStatusRecordTest
ReadAttributeStatusRecord [status=FAILURE, attributeIdentifier=1]
ReadAttributeStatusRecord [status=SUCCESS, attributeIdentifier=1, attributeDataType=ATTRIBUTEID, attributeValue=0]
Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.006 sec
Running com.zsmartsystems.zigbee.zcl.protocol.ZclCommandDirectionTest
[... cut ...]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :com.zsmartsystems.zigbee`

leonschenk, 2022-11-02:
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------< com.zsmartsystems.zigbee:com.zsmartsystems.zigbee >----------
[INFO] Building com.zsmartsystems.zigbee 1.4.10-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ com.zsmartsystems.zigbee ---
[INFO] Deleting E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\target
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.5:prepare-agent (jacoco-prepare-agent) @ com.zsmartsystems.zigbee ---
[INFO] argLine set to -javaagent:C:\\Users\\leon_\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.8.5\\org.jacoco.agent-0.8.5-runtime.jar=destfile=E:\\zsmarttest\\com.zsmartsystems.zigbee\\com.zsmartsystems.zigbee\\target\\jacoco.exec
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ com.zsmartsystems.zigbee ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\src\main\resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.8.0:compile (default-compile) @ com.zsmartsystems.zigbee ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 806 source files to E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\target\classes
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclFieldDeserializer.java: Some input files use or override a deprecated API.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclFieldDeserializer.java: Recompile with -Xlint:deprecation for details.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclFieldSerializer.java: Some input files use unchecked or unsafe operations.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclFieldSerializer.java: Recompile with -Xlint:unchecked for details.
[INFO] 
[INFO] --- maven-bundle-plugin:3.5.0:manifest (bundle-manifest) @ com.zsmartsystems.zigbee ---
[WARNING] Manifest com.zsmartsystems.zigbee:com.zsmartsystems.zigbee:jar:1.4.10-SNAPSHOT : Export com.zsmartsystems.zigbee,  has 1,  private references [com.zsmartsystems.zigbee.internal], 
[WARNING] Manifest com.zsmartsystems.zigbee:com.zsmartsystems.zigbee:jar:1.4.10-SNAPSHOT : Export com.zsmartsystems.zigbee.zcl,  has 1,  private references [com.zsmartsystems.zigbee.internal], 
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ com.zsmartsystems.zigbee ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\src\test\resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.8.0:testCompile (default-testCompile) @ com.zsmartsystems.zigbee ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 103 source files to E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\target\test-classes
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/test/java/com/zsmartsystems/zigbee/zcl/ZclClusterTest.java: Some input files use or override a deprecated API.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/test/java/com/zsmartsystems/zigbee/zcl/ZclClusterTest.java: Recompile with -Xlint:deprecation for details.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/test/java/com/zsmartsystems/zigbee/TestUtilities.java: Some input files use unchecked or unsafe operations.
[INFO] /E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/src/test/java/com/zsmartsystems/zigbee/TestUtilities.java: Recompile with -Xlint:unchecked for details.
[INFO] 
[INFO] --- maven-surefire-plugin:2.12.4:test (default-test) @ com.zsmartsystems.zigbee ---
[INFO] Surefire report directory: E:\zsmarttest\com.zsmartsystems.zigbee\com.zsmartsystems.zigbee\target\surefire-reports

-------------------------------------------------------
 T E S T S
-------------------------------------------------------
Running com.zsmartsystems.zigbee.app.basic.ZigBeeBasicServerExtensionTest
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 2.416 sec
Running com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtensionTest
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.zsmartsystems.zigbee.TestUtilities (file:/E:/zsmarttest/com.zsmartsystems.zigbee/com.zsmartsystems.zigbee/target/test-classes/) to field java.lang.reflect.Field.modifiers
WARNING: Please consider reporting this to the maintainers of com.zsmartsystems.zigbee.TestUtilities
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.844 sec
Running com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscovererTest
Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.312 sec
Running com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscovererTest
Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.346 sec
Running com.zsmartsystems.zigbee.app.iasclient.ZclIasZoneClientTest
[... cut ...]
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.apache.maven.surefire.util.ReflectionUtils.invokeMethodWithArray(ReflectionUtils.java:189)
	at org.apache.maven.surefire.booter.ProviderFactory$ProviderProxy.invoke(ProviderFactory.java:165)
	at org.apache.maven.surefire.booter.ProviderFactory.invokeProvider(ProviderFactory.java:85)
	at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:115)
	at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:75)

leonschenk, 2022-11-02:
@cdjackson here you see the logging from `mvn clean install > logfile.log`. It fails approximately once per 4 build cycles.
I build on windows.

cdjackson, 2022-11-02:
Thanks. I don't think I've ever seen this fail here and this code has been included in the library for a few years. It also hasn't ever failed with CI which runs on every PR and merge etc...

cdjackson, 2022-11-02:
It's also strange that the output from the error looks invalid -:

```
java.lang.AssertionError: expected: but was:
```

This can't be right since the line that reports to have a problem is this -:

```
        assertEquals(KeyEstablishmentState.UNINITIALISED,
                TestUtilities.getField(ZclKeyEstablishmentServer.class, keServer, "keyEstablishmentState"));

```

This shows that it is expected to be `UNINITIALISED`, so I don't know why the test output is printed as blank. 

If I change this test so it will fail, you can see it prints out the expected and actual values -:

```
java.lang.AssertionError: expected:<COMPLETE> but was:<UNINITIALISED>
	at org.junit.Assert.fail(Assert.java:89)
	at org.junit.Assert.failNotEquals(Assert.java:835)
	at org.junit.Assert.assertEquals(Assert.java:120)
	at org.junit.Assert.assertEquals(Assert.java:146)
	at com.zsmartsystems.zigbee.app.seclient.ZclKeyEstablishmentServerTest.InitiateKeyEstablishmentRequestCommandSuccess(ZclKeyEstablishmentServerTest.java:101)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

```

I've run just this test class loads more times and can't make it fail (other than if I change the expected result as above ;) ).

I wonder if there's something else going on with your environment given it seems to print invalid information. Maybe you can take a look at this if you get a chance?