zsmartsystems/com.zsmartsystems.zigbee #1360: Separate logs generated by multiple network managers in multiple files

mikomarrache, 2022-12-03:
We use the library with multiple Ember NCPs and currently we are not able to separate logs generated by network managers in multiple files and therefore debugging is very complicated.

Is there a way to differentiate logs generated by multiple instances with the actual code?

If not, which approach do you recommend? We would be happy to provide a PR for it but we first need to discuss the right solution.

cdjackson, 2022-12-03:
I don't think there's a way to differentiate this at the moment - at least not that I can think of off the top of my head.

I'm not against adding something to the logging - possibly adding the coordinator IEEE address in the logs would be best - probably at the beginning of each line. There might be some instances where this doesn't work (ie logging before the NCP has communication and we don't know the IEEE) but maybe you can live with this ambiguity? I think it's better to try and use something like the IEEE than to use some other key that might change each time the system is started, but I'm open to suggestions if you have some idea?

Are you wanting to differentiate all layers (ie EZSP and ASH)?

Let me check a couple of things here to see how big a pain it will cause by adding this on the logs. I have some processing software which does read these lines - I think it searches for certain keywords such as `EZSP RX` so putting something before this shouldn't cause any issue -:

eg 

```
            logger.debug("RX EZSP: {}", response);
```

to 

```
            logger.debug("{}> RX EZSP: {}", ieeeAddress, response);
```

I'm happy to discuss other suggestions though...

mikomarrache, 2022-12-03:
> I don't think there's a way to differentiate this at the moment - at least not that I can think of off the top of my head.
>
> I'm not against adding something to the logging - possibly adding the [...]

Another approach I thought of is to provide a tag when creating the network manager and using this tag all over the log statements. I think it is more flexible than using the coordinator IEEE address (that can still be the default when no tag is set).

> 
> Are you wanting to differentiate all layers (ie EZSP and ASH)?

Yes - these layers are also important for debugging.

> Let me check a couple of things here to see how big a pain it will cause by adding this on the logs. I have some processing software which does read these lines - I think it searches for certain [...]