zsmartsystems/com.zsmartsystems.zigbee #1361: Type of incoming/outgoing frame counters

mikomarrache, 2022-12-04:
The counters are declared as Integer here:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/b15130c021219cbfe138dc4ec60423a0b8b3f679/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ezsp/structure/EmberKeyStruct.java#L56

However, the specs says:

[image]
[image]
[image]

If I understand correctly, that may become an issue when the counters will reach high value, am I correct ?

If yes, should we declare the properties as Long instead of Integer ?

cdjackson, 2022-12-04:
Yes, values above 2^31 I guess will be an issue. It's not so obvious without looking at the code how to resolve this though as I'm not sure what impact changing this will have - not just on this class, but all the others that use `uint32_t`

cdjackson, 2022-12-04:
Just to add though that in reality this is unlikely to be an issue since 2^31 is still a very large number :)

At this rate, it would take around 7 years to overflow assuming 10 commands were sent every second - or 68 years if it's just 1 command per second. So the reality is that it's very (very!) unlikely to cause a problem IMHO.