zsmartsystems/com.zsmartsystems.zigbee #1362: ASH frame handler - receive queue is full

mikomarrache, 2022-12-07:
We encountered cases where the receiving queue of the ASH frame handler is full. Therefore, received frames are rejected (IllegalStateException : Queue full).

Here, we can see the queue capacity is set to 10. Also, the library doesn't expose methods to set the capacity during initialization.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/b15130c021219cbfe138dc4ec60423a0b8b3f679/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/ash/AshFrameHandler.java#L117

Is there a reason the capacity is set to 10? Do you think increasing the capacity could cause other issues?

Thanks

sorinpenteleiciuc, 2022-12-07:
I also subscribe to your ideea to increase this size to at least 100.

cdjackson, 2022-12-09:
I think it's useful to understand why the queue fills up before making any change. @sorinpenteleiciuc I believe when we saw this problem with your system some time back it was caused by other activities taking a lot of time on your low powered processor.

IMHO it's always a difficult issue, and increasing the queue size may cause other problems. The queue is meant to manage a real-time incoming data stream - queuing a lot of frames will mean that frames may now be in the queue for many seconds, and this may mean that we are replying to data that has already been retried by the sending device if it doesn't get a timely response. This just causes more problems, and IMHO we're better off not to increase the queue size to tater for such instances, but it's better to drop the frames.

As an example, if we were to increase the queue size to 100 as suggested here, that means that we now have up to 100 frames in the queue to process, and this will likely be impossible. The system can only send (from memory) 10 to 20 frames per second, so we have 5 to 10 seconds worth of data in the queue, but the timeout will be in the order of 5 to 7 seconds, so we have a problem. 

Thats my view - maybe I've missed something? And maybe 10 isn't the perfect number and I could be convinced to change it if there was a good rational - supported by logs and an understanding of why it's a good idea.