zsmartsystems/com.zsmartsystems.zigbee #1363: ArrayIndexOutOfBoundsException during shuffle of outstanding queues (transaction manager)

mikomarrache, 2022-12-08:
We got the following exception:

```
java.lang.ArrayIndexOutOfBoundsException: 3
    at java.util.concurrent.CopyOnWriteArrayList.get(CopyOnWriteArrayList.java:388)
    at java.util.concurrent.CopyOnWriteArrayList.get(CopyOnWriteArrayList.java:397)
    at java.util.Collections.swap(Collections.java:499)
    at java.util.Collections.shuffle(Collections.java:460)
    at java.util.Collections.shuffle(Collections.java:427)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:656)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.transactionComplete(ZigBeeTransactionManager.java:540)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.failTransaction(ZigBeeTransaction.java:446)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.transactionStatusReceived(ZigBeeTransaction.java:491)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager$3.run(ZigBeeTransactionManager.java:579)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
    at java.lang.Thread.run(Thread.java:748)
```

It happened only once but I think it may happen if the list is shuffled while it is modified.

cdjackson, 2022-12-09:
I thought that this shouldn't be possible since this is a `CopyOnWriteArrayList` so it should manage concurrent access -:

> A thread-safe variant of [ArrayList](https://docs.oracle.com/javase/7/docs/api/java/util/ArrayList.html) in which all mutative operations (add, set, and so on) are implemented by making a fresh copy of the underlying array.

Maybe something in here isn't safe (I guess given the OOB exception).  Also, I would have expected a CME if there was a concurrent modification problem?

cdjackson, 2022-12-09:
Thanks for the explanation - I'd not looked at how this was being used.

> Do we need the shuffle method to be called here?

Yes - at least we need some way to randomise the queues - otherwise we'd send all frames for one device before moving on to the next device. So while `shuffle` probably isn't required - some way to select a random queue is needed. Maybe there's a better option though than `shuffle`.

ViToni, 2023-03-22:
> Yes - at least we need some way to randomise the queues - otherwise we'd send all frames for one device before moving on to the next device. So while shuffle probably isn't required - some way to select a random queue is needed. Maybe there's a better option though than shuffle.

Is there a special reason which makes it better to have multiple queues to take work from compared to a common queue where outstanding work is put in and then all outstanding work is processed from there?

ViToni, 2023-03-23:
@cdjackson Thanks for claryfing. Maybe I can rephrase the question: can we uses a central queue to manage the transactions queue, thereby keeping order and covering the side cases like sleepy devices and "blocked" devices?
I created PR #1381 to clarify my approach. Maybe you could give it a look to see if the idea is feasible.

PR #1381 (2023-04-21): Rework queue management for concurrency
Changes the transaction queue handling for hopefully better concurrency.
Instead of iteration over transaction queues, each with their own order, there is now a queue of queues.
Each time a transaction is submitted for sending, its queue is added to the main queue resulting into an ordered sequence of transactions, indirectly represented by the queued transaction queues.
For each entry in the main queue only one transaction is processed. Sleepy queues and queues with a release time in the future will be readded to the main queue.

This should fix #1363 a [... cut]
closed 2023-04-21 by cdjackson