zsmartsystems/com.zsmartsystems.zigbee #1365: ConcurrentModificationException while iterating queues

mikomarrache, 2022-12-13:
We found another issue related to concurrency:

```
java.util.ConcurrentModificationException: null
    at java.util.ArrayDeque$DeqIterator.next(ArrayDeque.java:648)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue.shutdown(ZigBeeTransactionQueue.java:137)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.removeNode(ZigBeeTransactionManager.java:607)
    at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.nodeRemoved(ZigBeeTransactionManager.java:786)
    at com.zsmartsystems.zigbee.ZigBeeNetworkManager$10.run(ZigBeeNetworkManager.java:1689)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
    at java.lang.Thread.run(Thread.java:748)
```

We are investigating how to solve it.

mikomarrache, 2022-12-13:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/0c393dbb7df447522347ad5bfd7352a94dc68161/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/transaction/ZigBeeTransactionQueue.java#L69

Regarding the first exception, should we replace the `ArrayDeque` by `ConcurrentLinkedDeque` ?

cdjackson, 2022-12-13:
Agreed - I think using `ConcurrentLinkedDeque` probably makes sense here so please feel free to provide a PR.

mikomarrache, 2022-12-13:
Regarding the second exception, it seems that cancelling the transaction somehow modifies the outstandingTransactions set.

Since synchronized blocks are re-entrant, modifying the set while iterating is possible.

I'm not sure yet how to solve it though.

cdjackson, 2023-04-22:
This should be closed with #1375

closed 2023-04-22 by cdjackson