zsmartsystems/com.zsmartsystems.zigbee #1382: Should ZigBeeDongleTelegesis return NO_NETWORK when Network Information Command returns "NoPAN"?

TomTravaglino, 2023-04-18:
In `com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis.startup(boolean)` there is this code block used to check if the network is up
```
// Check if the network is now up
TelegesisDisplayNetworkInformationCommand networkInfo = new TelegesisDisplayNetworkInformationCommand();
if (frameHandler.sendRequest(networkInfo) == null || networkInfo.getStatus() != TelegesisStatusCode.SUCCESS) {
    return ZigBeeStatus.BAD_RESPONSE;
}
if (networkInfo.getDevice() != TelegesisDeviceType.COO) {
    return ZigBeeStatus.INVALID_STATE;
}
```

In `TelegesisDisplayNetworkInformationCommand.deserialize` there is such a check
```
if (device!=TelegesisDeviceType.NOPAN) {
    // Deserialize field "channel"
    channel = deserializeInteger();
    stepDeserializer();
...
}
``` 
My understanding of this is that when the device type is NOPAN, there is no valid network configured. Shouldn't this also be checked in `ZigBeeDongleTelegesis.startup` and then return ZigBeeStatus.NO_NETWORK? I mean something like this
```
TelegesisDisplayNetworkInformationCommand networkInfo ...
...
if (networkInfo.getDevice() == TelegesisDeviceType.NOPAN) {
    return ZigBeeStatus.NO_NETWORK;
}
...
```

cdjackson, 2023-04-18:
Hey @TomTravaglino - hope you're well. I agree - your proposal sounds fine to me if you want to create a PR and do a quick test as I'm not sure I have a Telegesis stick to hand at the moment...

TomTravaglino, 2023-04-18:
@cdjackson I'm fine, thank you. I hope you are well too. I will do some tests and then create the PR.

TomTravaglino, 2023-04-19:
@cdjackson I created the PR. Please take a look.

PR #1387 (2023-04-20): Make ZigBeeDongleTelegesis return NO_NETWORK when NoPAN
This PR makes ZigBeeDongleTelegesis return `ZigBeeStatus.NO_NETWORK` when the result of the "Display Product Identification" command is `NoPAN`.

Closes #1382
closed 2023-04-20 by cdjackson