zsmartsystems/com.zsmartsystems.zigbee #1390: Animal sniffer reports errors on projects

ViToni, 2023-04-22:
**Describe the bug**
Running `mvn animal-sniffer:check` on the root project of current `master` gives me animal sniffer errors in
* `com.zsmartsystems.zigbee.autocode`
* `com.zsmartsystems.zigbee.console`
* `com.zsmartsystems.zigbee.console.main`

**To Reproduce**
`mvn animal-sniffer:check` within roo project

**Expected behavior**
The check should run without issues.

**Additional context**
The error only goes away when switching to at least:
```
<signature>
	<groupId>net.sf.androidscents.signature</groupId>
	<artifactId>android-api-level-26</artifactId>
	<version>8.0.0_r2</version>
</signature>
```
on the root project. (According to https://apilevels.com/ Android Oreo is from 2017.)

The other option is to move the configuration
```
<signature>
	<groupId>net.sf.androidscents.signature</groupId>
	<artifactId>android-api-level-21</artifactId>
	<version>5.0.1_r2</version>
</signature>
```
from the root project to `com.zsmartsystems.zigbee`.

I can open a PR for either solutions, whatever the prefered approach is (I'm missing too much background about how all projects work together to make a proposal for either of the approaches.)

----

While testing for the minimal version which does not report an error I saw that there are two versions of the animal sniffer plugin used. There is now PR #1389 to align these versions (does not change the signature check).

cdjackson, 2023-04-22:
I need to have a look at this, but I've not noticed any errors in the build and am surprised that animal sniffer is enabled in these projects. These three projects shouldn't really be an issue though as they are not really applicable to Android. The code generators only run on the PC, and likewise the console is only applicable to the PC.

Let me take a look...

cdjackson, 2023-04-22:
I didn't get any errors on the autocoder, but I've in any case skipped animal sniffer on the three projects you mention above. I think this should resolve the issue here since as above, these projects are not applicable for Android.

We have Android users using older versions of Android still (AFAIK) and I'm hesitant to update to a newer minimum version and break stuff unnecessarily...

closed 2023-04-22 by cdjackson

ViToni, 2023-04-22:
Thanks for clarifying on that.
Strange that you didn't encounter the errors, I'll double check my environment.

PR #1391 (2023-04-22): Remove animal-sniffer on some projects
Animal sniffer is not applicable on some projects - ie those that are not compatible with Android. Typically this are the console application, and the autocode generators.

Closes #1390