zsmartsystems/com.zsmartsystems.zigbee #1401: EZSP v12 (EmberZNet Serial Protocol versions 12) for Silicon Labs Ember NCP dongles?

Hedda, 2023-06-09:
**Is your feature request related to a problem? Please describe.**

EZSP protocol version been incremented to EZSP v12 in Silicon Labs Gecko SDK v4.3 (4.3.0.0) / Zigbee EmberZNet v7.3 (7.3.0.0):

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.0.0.pdf

https://www.silabs.com/wireless/zigbee?tab=software#software

0x0C hexadecimal = 12 in decimal.

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.3/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

Silicon Labs EmberZNet 7.3.0.0 makes Silabs  Zigbee stack compatible Zigbee Protocol Specification Revision 23 / R23 (Zigbee PRO 2023), but users should be warned against flashing EZSP v12 firmware as of yet as it is not backwards compatible with EZSP v11 or older EZSP.

**Zigbee**
- Zigbee R23 compliance, with these Security enhancements among others:
  - Dynamic link key negotiation
  - Device interview to query devices before they are allowed to join
  - Trust Center Swap Out to replace an existing Trust Center with a new one
  - Frame Counter Synchronization
- Zigbee Direct Device (ZDD) support for:
  - Onboarding/commissioning
  - Communication to all Zigbee devices without a hub (Alpha), using Bluetooth LE
- Zigbee Smart Energy 1.4a compliance (Alpha)
- Enhancements to Zigbee Green Power (ZGP) APIs
- New Zigbee Security upgrade component for moving encryption keys from cleartext NVM3 tokens into secure storage

**Multiprotocol**
- Zigbee/OpenThread Concurrent Multiprotocol SoC sample app
- CPC GPIO expander module
- Zigbeed enhancements

https://www.silabs.com/blog/zigbee-pro-2023-spec-released-increases-security

https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

**Describe the solution you'd like**

Support for EZSP v12 (EmberZNet Serial Protocol versions 12) for Silicon Labs Ember NCP dongles.

**Describe alternatives you've considered**

N/A.

**Additional context**

Connectivity Standards Alliance (CSA) have now ratified and finally officially released Zigbee Pro Specification Revision 23 / R23 (a.k.a. Zigbee PRO 2023) on March 15th of 2023 as the very latest latest Zigbee specification standard:

- https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

This public press release announcement of Zigbee PRO 2023 by CSA was accompanying these official news blog articles:

- https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

**_Zigbee PRO 2023 introduces the coexistence of Zigbee and Zigbee Smart Energy devices, adds SubGHz support to North American and EU markets, and offers device onboarding and control via Bluetooth Low Energy._**

Though "Zigbee Direct 1.0" was stealth launched in advance already back in January of 2023 it is also part of Zigbee PRO 2023:

- https://csa-iot.org/newsroom/zigbee-direct/

- https://csa-iot.org/wp-content/uploads/2022/12/20-27688-037-zigbee_direct_spec.pdf

FYI, about a year ago CSA (formerly Zigbee Alliance) also posted a blog on Zigbee features related to Zigbee Revision 23 (R23):

* https://csa-iot.org/newsroom/looking-ahead-industry-trends-and-new-zigbee-features/

[... 17 lines cut ...]
- _**Device Interview** - Device Interview technology enables users to query and filter devices before allowing them onto a network based on ecosystem requirements. Also exchanges encrypted messages between the Trust Center and joining devices to vet unauthorized users. This process allows for secure firmware upgrades and detailed device filtering as new entities enter a network_
- _**Zigbee Direct** can simplify device onboarding (and control) utilizing Bluetooth Low Energy, a technology found in many voice assistant products and mobile devices. Uses new virtual devices **Zigbee Direct Device (ZDD)** and **Zigbee Virtual Device (ZVD)**._
  - https://csa-iot.org/newsroom/the-connectivity-standards-alliance-introduces-zigbee-direct-simplifying-integration-with-bluetooth-low-energy-devices/
    - https://www.youtube.com/watch?v=OYgbLTA2eq8&ab_channel=ConnectivityStandardsAlliance
    - https://www.youtube.com/watch?v=py2O0OJwick&ab_channel=ConnectivityStandardsAlliance
    - _**Note!** Starting with Zigbee PRO R23 specification onwards, Zigbee Coordinators are required to implement a minimum set of Zigbee Direct support features, such as handling Zigbee Direct-specific devices and key types. These requirements are extensions to the Zigbee stack as run on top of 802.15.4. Coordinators are NOT required to implement Bluetooth Low Energy._
        - https://csa-iot.org/all-solutions/zigbee/zigbee-direct/
          - https://csa-iot.org/all-solutions/zigbee/zigbee-direct-faq/
            - https://csa-iot.org/newsroom/connectivity-standards-alliances-zigbee-direct-feature-in-development/
            -   https://www.ubisys.de/en/company/ubisysblog-aboutubisys/zigbee-direct-golden-units/
            -   https://www.telink-semi.com/zigbee-direct/
            -   https://www.silabs.com/support/training/zgb-201-what-is-zigbee-direct
            -   https://pt.dsr-corporation.com/news/dsr-supports-zigbee-direct/
              -   https://pt.dsr-corporation.com/news/how-dsr-zigbee-direct-solutions-impact-your-users/
                - https://www.youtube.com/watch?v=TlCX95jTrWQ&ab_channel=DSRCorporation
- _**Regional SubGHz support extends the supported bands beyond the 2.4 GHz range** - Adds the support for sub-gigahertz frequencies in Europe (800 MHz) and North America (900 MHz) and enhances signal strength and range, accommodating even more use cases._
- _**Trust Center Swap-Out** - Feature permits the replacement of the 'Trust Center' - which can be a gateway, hub, smart speaker, or even commercial electric meters - for a network without necessitating the recommissioning of all devices._
- _**Restricted Mode**-  Limits the ability for networks to change the state of any device and, instead, allocates that role to the Trust Center only. This blocks unauthorized device changes and gives managers better visibility and control over device configuration._
- _**Dynamic Link Key (DLK)** - The Dynamic Link Key, based on Public/Private key pairing and advanced security curves, provides further protection against network attacks._
  - https://csa-iot.org/newsroom/zigbee-feature-dynamic-link-key/
    - https://www.youtube.com/watch?v=YNGVyQNgJX4&ab_channel=ConnectivityStandardsAlliance
- _**APS Frame Counter Synchronization** - Detects seemingly legitimate messages that are actually duplicates used in a replay attack, where an adversary sends an intercepted message back to gain access. APS frame counter synchronization validates all messages between endpoints to differentiate real messages from false ones._
- _**Secured Channel and PAN ID Changes** -  PAN ID identifies a particular Zigbee network, while Secured Channel is a chosen channel maintained and selected by the Trust Center. Previously, any network endpoint could report busy channels, which in turn can disrupt network operations. To prevent unnecessary changes, Zigbee R23 deploys additional measures to ensure that only network managers have this power._

PS: Texas Instruments Z-Stack ZNP SimpleLink SDK has already since a long time been updated to Zigbee R23, as TI has been an early promoter member of the Connectivity Standards Alliance (formerly Zigbee Alliance) for more than 12 years and is one of CSA's lead participant members in Zigbee R23 (and Matter) standards as well as being the chair for CSA's Zigbee Sub-1G task group.

Hedda, 2023-06-20:
FYI, Nabu Casa already offers official EmberZNet 7.3 NCP firmware beta images for the [Home Assistant SkyConnect](https://www.home-assistant.io/skyconnect/) USB dongle:

https://github.com/NabuCasa/silabs-firmware/tree/main/EmberZNet/beta

[darkxst](https://github.com/darkxst)'s has also added support for multiple hardware manifests to his fork of Nabu Casa's silabs-firmware-builder in order to allow override configs for automatic builds to enable automatic building of (unofficial) firmware images for many different Silicon Labs based EFR32 adapters, including; , ITead Sonoff ZBDongle-E, Easyiot ZB-GW04, and Elelabs ELR023 + ELU013 :

https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds

Note! He also has experimental builds for same EmberZNet version and hardware but with different baud rate speeds for testing.

https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests

FranzSchi, 2023-08-30:
One more smal hint. I'm using this Library with the openhab plugion.
The Message shown is:
`EZSP Dongle: NCP requires unsupported version of EZSP (required = V12, supported = V4)`
I guess "supported = V4" should be upgraded to V11 or V12 in this context.
Thx
/Franz

cdjackson, 2023-08-30:
> I guess "supported = V4" should be upgraded to V11 or V12 in this context.

No - this number is not a static - it is a value reported by the dongle.

FranzSchi, 2023-08-31:
Yes, exactly the suppoted shall be upgraded from  4 to 11 or if this is implemented to 12

cdjackson, 2023-08-31:
> I guess that No modern stick has V4 installed.

Well, I have plenty here that I use regularly. There's nothing wrong with the older EM357 devices for use with Zigbee. Anyway, this is just the minimum version in this case, so it's irrelevant.

Anyway, it looks like the log reporting is around the wrong way - so it should say supported = V12, required = V4.

FranzSchi, 2023-09-01:
Hi @cdjackson,

there is noting to say against V4 but it seems I'm not the only one that misunderstood it.
I would suggest to use both like:
EZSP Dongle: NCP requires unsupported version of EZSP (yours= V12, minsupported = V4, maxsupported= V11)

[image]

Source: https://github.com/xsp1989/zigbeeFirmware/issues/44#issuecomment-1523466896

BR
/Franz

ViToni, 2023-11-23:
Is the core issue resolved by #1410? (Even if it doesn't solve the misleading log message yet.)

BTW I like @FranzSchi's suggestion a lot about how to phrase the log message. At least vor me it would be much clearer what is going on.

cdjackson, 2023-11-23:
Not all new features added to the Silabs firmware have been implemented, but yes, EZSP12 is implemented.

IIRC the log message was also changed a little while ago as suggested (from memory).

So I think both features are fundamentally included in the latest release even if there are more unimplemented features available in the firmware, they are not currently utilised by the library. So I think I will close this and if there's something specific required, we should open a new issue specifically for that.

closed 2023-11-23 by cdjackson

ViToni, 2023-11-23:
The log message was changed by #1409 to: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c3a95652adf4c681fb60f36449e2ee3b894e84d6/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L1331-L1332

Looks great.

Hedda, 2023-11-24:
> [darkxst](https://github.com/darkxst)'s has also added support for multiple hardware manifests to his fork of Nabu Casa's silabs-firmware-builder in order to allow override configs for automatic builds to enable automatic building of (unofficial) firmware images for many different popular Silicon Labs EFR32 based Zigbee adapters, including; ITead Sonoff ZBDongle-E, Easyiot ZB-GW04, and Elelabs ELR023 + ELU013 :
> 
> https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds
> 
> Note! He also has experimental builds for same EmberZNet version and hardware but with different baud rate speeds for testing.
> 
> https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests

Off-topic but a FYI tip for devs, darkxst is now providing prebuild cutting-edge firmware for most common EFR32MGxx adapters:

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs-SL/firmware_builds

`ncp-uart-hw-` = EZSP mode (pure EmberZnet Zigbee NCP).
`rcp-uart-802154-` = RCP mode for MultiPAN/multiprotocol (requires Silicon Labs EZSP and/or OpenThread daemon on the host).
`ot-rcp-` = OpenThread Only (Thread Spinel mode).

Web Flasher - Flash newer Silabs EFR32 adapters directly from your browser (only Chrome and Edge supported) -> [SL Web Flasher](https://darkxst.github.io/silabs-firmware-builder/)

Read this [blog post](https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware) for more details and instructions for using RCP Multi-PAN firmware (with Silabs Multiprotocol Addon for HA).