zsmartsystems/com.zsmartsystems.zigbee #1404: Insecure Zigbee Network Encryption Key Generation

0xbb, 2023-08-17:
I seems to me that this projects uses the insecure `Math.random` method to generate the Zigbee encryption key:  https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/61460438ce445b7d8fa45a055cf00082d8e68c8e/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/security/ZigBeeKey.java#L247

`SecureRandom` should be a secure replacement for the current method.

Also see https://github.com/Koenkk/zigbee2mqtt/issues/18313 for the same mistake

-----

I think users should regenerate existing network keys to be secure. It maybe also be good to request a CVE for this issue.

cdjackson, 2023-08-19:
The key will still be random unless the same seed is used multiple times, so I don't think there is really an issue here. From my reading of the difference, the secure version generates a larger number (128 bits vs 48 bits), but as we are only generating numbers between 0 and 255, that doesn't really matter.

So, while I don't think there's a significant security issue here, I do agree it would be better to use the new version and it would be good if you wanted to provide a PR?

0xbb, 2023-08-19:
I don't want to provide a PR, but below you find the new variant of the the code:

```java
...
import java.security.SecureRandom;
...

    /**
     * Create a {@link ZigBeeKey} with a random key
     *
     * @return {@link ZigBeeKey} containing a random 128 bit key
     */
    public static ZigBeeKey createRandom() {
        SecureRandom secureRandom = new SecureRandom();
        int key[] = new int[16];
        for (int cnt = 0; cnt < 16; cnt++) {
            key[cnt] = secureRandom.nextInt(256);
        }

        return new ZigBeeKey(key);
    }
```

cdjackson, 2023-08-20:
Thanks. I understand the code change, but thought if you could provide a PR that would be useful as I’m currently travelling for the next few weeks and can’t really do so…Sent from my iPhone

     * Create a ***@***.*** ZigBeeKey} with a random key
     *
     * @return ***@***.*** ZigBeeKey} containing a random 128 bit key
     */

PR #1408 (2023-10-25): Change random key generator to use secure random generator
Closes #1404
closed 2023-10-25 by cdjackson