zsmartsystems/com.zsmartsystems.zigbee #1414: Console doesn't start - CC2531

MorganMalavasi, 2024-01-10:
Hi, i'm trying to build some java code to backup the dongle CC2531, in particular the state of the network. If a lose the key or it breaks, I want to restore it without making the pairing step of each device again.

I've spotted this library, I want to try the console as explained in the doc because it contains maybe the script to do what I want, so without rewriting everything by myself, but I'm not able to run it. When I start the console it breaks because the library sends some unknown packet to the dongle.

Here the parameters i'm passing "-dongle CC2531 -port COM22 -baud 115200 -flow none"

Here my output logs:

Initialising ZigBee console...
17:07:09.423  DEBUG  Default: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=1, interTransactionDelay=50, maxRetries=2]
17:07:09.423  DEBUG  Broadcast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=4000, maxRetries=0]
17:07:09.423  DEBUG  Multicast: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=3, interTransactionDelay=1200, maxRetries=0]
17:07:09.566  DEBUG  ZigBeeNetworkManager initialize: networkState=UNINITIALISED
17:07:09.566  DEBUG  Network state is updated to INITIALISING
ZigBee network state updated to INITIALISING
17:07:09.569  DEBUG  CC2531 transport initialize
17:07:09.569  DEBUG  Opening port COM60 at 115200 baud with FLOWCONTROL_OUT_NONE.
17:07:10.078  DEBUG  ->  SYS_RESET (Packet: subsystem=null, length=1, apiId=41 00, data=FE 01 41 00 01 41, checksum=41, error=false)
17:07:10.084  DEBUG  Discarded stream: expected start byte but received 00
17:07:11.654  DEBUG  <-- SYS_RESET_RESPONSE (FE 06 41 80 00 02 01 02 07 01 C0)
17:07:11.654  DEBUG  Received Async Cmd: Packet: subsystem=null, length=6, apiId=41 80, data=FE 06 41 80 00 02 01 02 07 01 C0, checksum=C0, error=false
17:07:11.655  DEBUG  ->  SYS_VERSION (Packet: subsystem=null, length=0, apiId=21 02, data=FE 00 21 02 23, checksum=23, error=false)
17:07:11.661  DEBUG  <-  SYS_VERSION_RESPONSE (FE 0A 61 02 02 01 02 07 01 14 64 34 01 00 2B)
17:07:11.661  DEBUG  CC2531 version is Software=2.7 Product=1 Hardware=1 Transport=2
17:07:11.662  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 01 20, checksum=20, error=false)
17:07:11.667  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:12.666  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:12.666  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 1]
17:07:13.682  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 01 20, checksum=20, error=false)
17:07:13.686  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:14.698  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:14.698  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 2]
17:07:15.717  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 01 20, checksum=20, error=false)
17:07:15.722  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:16.726  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:16.726  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 3]
17:07:16.726  WARN   Failed getDeviceInfo for 1 due to null value
networkManager.initialize returned SUCCESS
17:07:16.728  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false)
17:07:16.733  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:17.729  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:17.729  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 1]
17:07:18.741  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false)
17:07:18.746  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:19.751  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:19.751  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 2]
17:07:20.765  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 06 27, checksum=27, error=false)
17:07:20.770  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:21.765  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:21.765  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 3]
17:07:21.765  WARN   Failed getDeviceInfo for 6 due to null value
PAN ID          = 65535
17:07:21.765  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
17:07:21.770  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:22.775  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:22.775  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 1]
17:07:23.777  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
17:07:23.781  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:24.791  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:24.791  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 2]
17:07:25.804  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 07 26, checksum=26, error=false)
17:07:25.809  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:26.812  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:26.812  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 3]
17:07:26.812  WARN   Failed getDeviceInfo for 7 due to null value
Extended PAN ID = 0000000000000000
17:07:26.815  DEBUG  ->  ZB_GET_DEVICE_INFO (Packet: subsystem=null, length=1, apiId=26 06, data=FE 01 26 06 05 24, checksum=24, error=false)
17:07:26.821  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 06 42)
17:07:27.824  DEBUG  ZB_GET_DEVICE_INFO executed and timed out while waiting for response.
17:07:27.824  DEBUG  Failed to send ZB_GET_DEVICE_INFO [attempt 1]
[... 102 lines cut ...]
17:07:31.957  DEBUG  Adding supported server cluster 0B05
17:07:31.957  DEBUG  ClusterMatcher adding server cluster 0B05
17:07:31.957  DEBUG  ZigBeeNetworkManager startup: reinitialize=false, networkState=INITIALISING
17:07:31.957  DEBUG  CC2531 transport startup
17:07:31.958  DEBUG  Creating network as Coordinator
17:07:31.958  DEBUG  ->  ZDO_MSG_CB_REGISTER (Packet: subsystem=null, length=2, apiId=25 3E, data=FE 02 25 3E FF FF 19, checksum=19, error=false)
17:07:31.962  DEBUG  <-  ZDO_MSG_CB_REGISTER_SRSP (FE 01 65 3E 00 5A)
17:07:31.963  DEBUG  ->  ZB_WRITE_CONFIGURATION (Packet: subsystem=null, length=3, apiId=26 05, data=FE 03 26 05 8F 01 01 AF, checksum=AF, error=false)
17:07:31.967  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 05 41)
17:07:32.973  DEBUG  ZB_WRITE_CONFIGURATION executed and timed out while waiting for response.
17:07:32.973  DEBUG  Failed to send ZB_WRITE_CONFIGURATION [attempt 1]
17:07:33.985  DEBUG  ->  ZB_WRITE_CONFIGURATION (Packet: subsystem=null, length=3, apiId=26 05, data=FE 03 26 05 8F 01 01 AF, checksum=AF, error=false)
17:07:33.990  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 05 41)
17:07:34.991  DEBUG  ZB_WRITE_CONFIGURATION executed and timed out while waiting for response.
17:07:34.991  DEBUG  Failed to send ZB_WRITE_CONFIGURATION [attempt 2]
17:07:36.018  DEBUG  ->  ZB_WRITE_CONFIGURATION (Packet: subsystem=null, length=3, apiId=26 05, data=FE 03 26 05 8F 01 01 AF, checksum=AF, error=false)
17:07:36.022  DEBUG  <-  SYS_RPC_ERROR (FE 03 60 00 01 26 05 41)
17:07:37.019  DEBUG  ZB_WRITE_CONFIGURATION executed and timed out while waiting for response.
17:07:37.019  DEBUG  Failed to send ZB_WRITE_CONFIGURATION [attempt 3]
17:07:37.019  ERROR  Failed to start zigbee network.
17:07:37.020  INFO   Serial port 'COM60' closed.

I don't know if this is a bug or I'm missing something. In this second case I ask you if you a have a solution.

Thanks

cdjackson, 2024-01-26:
I'm reasonably sure that the issue here is a compatability with the firmware in your stick. Your device reports version 2, and the library currently required V1. There is a PR in #1352 that is looking to implement V2, but it's currently stale. You might like to try that.

cdjackson, 2024-05-18:
Closing this due to inactivity

closed 2024-05-18 by cdjackson