zsmartsystems/com.zsmartsystems.zigbee #1452: Add support for Electrical Measurement (0x0B04) and Smart Energy Metering (0x0702) clusters for LEDVANCE Smart Plugs

amitjoy, 2024-12-10:
**Is your feature request related to a problem? Please describe.**
I’m currently unable to access energy monitoring features of LEDVANCE Smart+ ZigBee plugs because the `Electrical Measurement Cluster` (`0x0B04`) and the `Smart Energy Metering Cluster` (`0x0702`) are not supported by `com.zsmartsystems.zigbee.ZigBeeDeviceType`. These clusters are crucial for retrieving power consumption and energy usage data.

**Describe the solution you'd like**
I would like the ZSmartSystems ZigBee library to support the following clusters:
1. **Electrical Measurement Cluster** (`0x0B04`) – for measuring active power, voltage, and current.
2. **Smart Energy Metering Cluster** (`0x0702`) – for tracking cumulative energy consumption.

Adding these clusters will allow LEDVANCE Smart+ plugs and similar devices to report real-time power usage and cumulative energy data within openHAB.

**Describe alternatives you've considered**
NA

**Additional context**
- **Device**: LEDVANCE Smart+ Plug (ZigBee version)
- **Manufacturer**: LEDVANCE
- **Clusters to be supported**:
        - **Electrical Measurement Cluster** (**0x0B04**)
              - Active Power (0x050B)
              - RMS Voltage (0x0505)
              - RMS Current (0x0508)
	- **Smart Energy Metering Cluster** (**0x0702**)
	      - Current Summation Delivered (0x0000)
	      - Instantaneous Demand (0x0400)

This feature would allow users to fully utilize energy monitoring capabilities within openHAB, enhancing automation and energy management.

cdjackson, 2024-12-10:
Strange - these clusters have been supported for a very long time. We've even had a smart energy product go through certification.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/clusters/ZclElectricalMeasurementCluster.java

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/clusters/ZclMeteringCluster.java

You mention openHAB - some attributes of these clusters are already supported (for a long time) in openHAB - this includes the attributes that you have mentioned above (I think maybe RMS voltage isn't implemented in openHAB, but I believe all the others are, and have been for many years).

cdjackson, 2024-12-11:
Probably the console has added support for these clusters - it is certainly limited. You'd need to add them as locally supported clusters.