zsmartsystems/com.zsmartsystems.zigbee #1457: GitHub reports latest version to be 1.2.8.2 instead of 1.4.16.1

ViToni, 2025-02-03:
**Describe the issue**
Not a code bug but an issue regarding the organization of release version.
Looking on the project page GitHub reports  v1.2.8.2 to be the latest version instead of (as of now) [zigbee-1.4.16.1](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/releases/tag/zigbee-1.4.16.1)

**To Reproduce**
Steps to reproduce the behavior:
- Check the frontpage of this project 
[image]

- or visit https://github.com/zsmartsystems/com.zsmartsystems.zigbee/releases
[image]

**Expected behavior**
It would be nicer to have the latest version directly visible instead of having to go to [tags](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/tags).

cdjackson, 2025-02-03:
I do agree this is a bit annoying - I normally do all releases through maven, and I think I would need to manually go through Github release to make this show up on github.

There is an indicator of the currently released version in maven central slightly further down the page that's easy to see without having to go to the tags page -:

[image]

ViToni, 2025-02-04:
My bad. I hadn't scrolled to the README and missed the maven central badge. Thanks for clarifying @cdjackson.

closed 2025-02-04 by ViToni